marek-simonik/record3d #2: Segfault when streaming from iPad

jankais3r, 2020-10-06:
Hi Marek,

I am working on an app that uses the USB streaming feature of your app. The Python Demo App works flawlessly with my iPhone, however whenever I try to use it with iPad Pro 2020 I get a segfault and Python crashes.

I am attaching several crash logs produced this way. It happens in 100% of cases with my iPad.

macOS: 10.14.6
iPadOS: 13.5

```
$ python3 demo-main.py
Searching for devices
1 device(s) found
	ID: 4779
	UDID: 00008027-001E091A3606802E

Segmentation fault: 11
```

[Crash.zip](https://github.com/marek-simonik/record3d/files/5335955/Crash.zip)

marek-simonik, 2020-10-06:
Hi Jan,

I suspect you have iOS 14 installed on your iPhone, am I right? The Python crashes are most probably caused because you do not have the latest version of Record3D installed on your iPad. The latest version of Record3D can only be installed when you are running iOS/iPadOS 14.

You have two options how to fix the crashing:

1. (**Recommended**) Update to iPadOS 14 and download the latest Record3D update. The post-iOS-14 version of the library allows you to stream not only using the selfie FaceID camera, but also using LiDAR. Due to addition of LiDAR support, the library differs from its pre-iOS-14 version.
2. Stay on iPadOS 13.5. Then you will need to clone the [`version/1.0.1` branch](https://github.com/marek-simonik/record3d/tree/version/v1.0.1) (i.e. `git clone -b version/v1.0.1 https://github.com/marek-simonik/record3d.git`) and build the library from sources.

jankais3r, 2020-10-07:
Thank you, that explains it.

closed 2020-10-07 by jankais3r