marek-simonik/record3d #6: How to get the pointcloud from the USB-streamed depth images

j96w, 2020-11-06:
Hi, many thanks for this great project. I'm looking for a way to transform the USB-streamed depth images into a pointcloud on my pc, just like the results shown in the app. Could you share the code of how you make the transformation from depth to pointcloud? Thanks!

j96w, 2020-11-06:
Oh, I just found your intrinsic_mat function in the demo-main.py. Now I can transform the depth to pointcloud with that. Thanks!

closed 2020-11-06 by j96w

hetryn, 2021-07-25:
Hi, apologies for hoping onto this thread. It's my first time playing with this app and 3D images and hoping to get some help.

I have a in my directory some `/*.jpg` and `/*.depth` files. How do I use the `get_intrinsic_mat_from_coeffs()` function to generate point clouds. Should I be looking at some files with coeff values as well?

marek-simonik, 2021-07-26:
Hi,
the four coefficients are `fx, fy, tx, ty` (in that order). Please see [this section](https://github.com/marek-simonik/record3d_unity_streaming/blob/49152244daa10686220c06c4f80dd9a823357e0f/Record3DUnityStreaming.cpp#L72) of the Unity streaming library to see how a 3D point is generated using those parameters.

[Here](https://github.com/marek-simonik/record3d/issues/7#issuecomment-736573410) you can see a Python script that shows you how to load a `.depth` file.

P.S. I might be slow to respond on Github, so feel free to send me an email at [support@record3d.app](mailto:support@record3d.app).