marek-simonik/record3d #8: What ARkit API are you using?

Tetsu5902, 2020-12-19:
Hi,

What ARkit API are you using to get the depth, and is there any post-processing on the record3D app side for the depth?

Tetsu5902, 2020-12-22:
I just understood from [this video](https://developer.apple.com/videos/play/wwdc2020/10611) that it's sceneDepth API.
Thanks!

closed 2020-12-22 by Tetsu5902

marek-simonik, 2020-12-23:
Hi,
I apologise for the late reply. You are correct about the API. Record3D does not apply post-processing when streaming via USB, because I wanted every developer to apply their own post-processing, which fits the most to their needs.

andybak, 2020-12-23:
> Record3D does not apply post-processing when streaming via USB, 

Is that different to the data saved to .r3d?

I'm writing some code that I'm hoping will be interchangable between streaming and decoded r3d files. Do I need to handle depth any differently in either case?

marek-simonik, 2020-12-23:
> Is that different to the data saved to .r3d?

No, at least not yet. Both the USB live streaming and the `.r3d` files contain raw (i.e. *not* post-processed) depth data at this moment.

Tetsu5902, 2020-12-29:
Thank you for your reply!

>Record3D does not apply post-processing when streaming via USB, because I wanted every developer to apply their own post-processing, which fits the most to their needs.

Excellent work!!

marek-simonik, 2020-12-29:
Thank you, @Tetsu5902 :)!