marek-simonik/record3d #9: Error in demo.exe

vrtx-voiod, 2020-12-23:
I've successfully built the solution, but when I run the demo.exe I get this error:

[image]

marek-simonik, 2020-12-23:
Judging from the (partially occluded) last line in the command line window, it seems like the crash might have something to do with a mutex in the code.

I suspect it could relate to [this mutex](https://github.com/marek-simonik/record3d/blob/master/src/DemoMain.cpp#L128).

Try to replace this whole portion of code:

```cpp
#if WIN32
    std::recursive_mutex mainThreadLock_{};
#else
    std::mutex mainThreadLock_{};
#endif
```

by this:

```cpp
std::recursive_mutex mainThreadLock_{};
```

There were issues with the regular `std::mutex` on Windows, so I replaced it by `std::recursive_mutex`. It may be possible that `WIN32` is not `#define`d in your environment, in which case the ordinary mutex would be used. By explicitly using `std::recursive_mutex mainThreadLock_{};`, we can see whether that was really the cause of the error. Let me know if that helped or not.

vrtx-voiod, 2020-12-23:
Unfortunately it didn't work, it's still the same error:

`D:\a01\_work\6\s\src\vctools\crt\github\stl\src\mutex.cpp(167): unlock of unowned mutex`

marek-simonik, 2020-12-23:
That is interesting — I want to reproduce the error, but I cannot on my machine. May I ask which compiler do you use — is it MSVC or a different one?

By the way, I am just curious, does this error happen also when you run the demo in Release instead of Debug?

vrtx-voiod, 2020-12-24:
- I am using Visual Studio Community16.8.3 2019
- for CMake I use the GUI Version 3.19.1 where I set the OpenCV_DIR to my vc15 lib folder since I could not figure out how to set this up properly inside the console
- I set my architecture to x64 during CMake and final VS build

After starting from scratch I have realized that during build 2 parts are skipped:

`1>------ Skipped Build: Project: INSTALL, Configuration: Debug x64 ------
1>Project not selected to build for this solution configuration 
2>------ Skipped Build: Project: ALL_BUILD, Configuration: Debug x64 ------
2>Project not selected to build for this solution configuration 
========== Build: 0 succeeded, 0 failed, 6 up-to-date, 2 skipped ==========`

I also set up a solution for Visual Studio Community 2017 (2) 15.9.24 and get the very same error message:

`D:\a01\_work\6\s\src\vctools\crt\github\stl\src\mutex.cpp(167): unlock of unowned mutex`

Sometimes i addition this error occurs in the debug console:

`Unhandled exception at 0x00007FFCAEA274BB (opencv_world412d.dll) in demo.exe: 0xC0000005: Access violation reading location 0x0000000000000000. occurred`

But I realized that this only happens when I did not start the record in the iOS app.

marek-simonik, 2020-12-25:
I have just merged a pull request by @y-labo ([https://github.com/marek-simonik/record3d/pull/10](https://github.com/marek-simonik/record3d/pull/10)). Please download the latest [`DemoMain.cpp`](https://github.com/marek-simonik/record3d/blob/master/src/DemoMain.cpp) to see whether the error you encountered has been fixed.

marek-simonik, 2021-01-29:
@vrtx-voiod since you gave thumbs up to my latest message and have not reported a new issue with the demo, I assume the problem was solved, so I am closing this issue.

closed 2021-01-29 by marek-simonik