marek-simonik/record3d #18: Strange LiDAR intrinsics

blackccpie, 2021-07-01:
Hi,

I have a strange behavior with my iPad Pro when inspecting the intrinsics of LiDAR frames I receive through USB streaming: the fx, fy, tx & ty parameters are not constant over time, they change depending on the scene... which I find quite strange. Is that a normal behavior?
Also, the frame resolution is 192x256, and the tx/ty values are around 366 and 476 respectively, which is also strange because the should not exceed frame resolution. Am I missing something?

Thanks for your help,

Regards

Albert

marek-simonik, 2021-07-28:
Hi Albert,

thank you for reporting this bug, it has been fixed now — please update to Record3D 1.6 or newer and download the newest version of the `record3d` library.

In the newest update of Record3D and the `record3d` library, the provided intrinsic parameters are the intrinsic parameters of the RGB image. In case you would stream with the "Higher-quality LiDAR recording" option enabled (turn on/off in the Settings tab), you will need to either:

- scale-up the corresponding depth map to match the RGB image resolution (because the RGB image has higher resolution than the depth map) and use the provided intrinsic parameters
- or scale the RGB intrinsic parameters to arrive at the intrinsic parameters of the depth image (`depthIntrParams = (depthMap.width / rgbImg.width) * rgbIntrParams`)

Let me know in case it would not work for you.

Best regards,
Marek

closed 2021-07-28 by marek-simonik

blackccpie, 2021-08-25:
Hi Marek,

Thanks for the new release, everything is working fine now. As discussed with you, the non-constant intrinsics values may be related to camera autofocus. Do you think autofocus enable/disable could be an option included in the record3d app?

Regards

Albert

saurabheights, 2023-09-17:
@marek-simonik As per your comment above, shouldnt documentation for `get_intrinsic_mat` be
"Returns the intrinsic matrix of current rgb frame." instead of
"Returns the intrinsic matrix of current Depth frame."

See https://github.com/marek-simonik/record3d/blob/master/python-bindings/src/PythonBindings.cpp#L46