marek-simonik/record3d #21: usbmuxd_get_device_list: error opening socket!

lamhoson, 2021-08-09:
Any ideas for the error and how to fix it?  Would the system work with a USB hub or must plug into the computer directly ?

[image]

marek-simonik, 2021-08-09:
I've never seen similar error before, but since it is related to the low-level iDevice–PC communication and not the `record3d` library per se, I can only guess and recommend checking the following:

- If you are on Linux: have you installed the `libusbmuxd-dev` library?
- If you are on Windows (I guess this is the case): have you installed iTunes and made sure you can see your iPhone/iPad in iTunes?
- Have you tapped the "*Trust*" button in the "*[Trust This Computer](https://support.apple.com/en-us/HT202778)*" dialog on your iPhone/iPad?

I would indeed first try connecting your iPhone/iPad to your computer directly, just to see if this is causing the issue, but my bet is on one of the reasons mentioned above.

I'm curious if some of the above mentioned advice will help, please keep me updated. Thank you.

lamhoson, 2021-08-09:
Noted with thanks. I didn't install iTune but just plugin to the notebook computer.

lamhoson, 2021-08-09:
However i can see my photos in the iPhone.

marek-simonik, 2021-08-09:
The missing iTunes seems to be the cause of the problem :)! Even though you can see the photos in Windows Explorer, it is still necessary to install iTunes to install necessary libraries for low-level communication with your iPhone.

Please install iTunes, restart your computer and try the Python demo again (it should work then).

lamhoson, 2021-08-09:
Noted with thanks and it work.

However which computer folder stored the files captured?

marek-simonik, 2021-08-09:
I'm glad to hear installing iTunes fixed the problem. I am therefore closing this issue, but we can still continue in our discussion.

> However which computer folder stored the files captured?

The RGBD images received during the USB live streaming are not stored on the disk — each newly received frame is stored in RAM, it is displayed and then discarded. If you want to save the received images, you will have to do it yourself (e.g. using OpenCV).

closed 2021-08-09 by marek-simonik

lamhoson, 2021-08-09:
Noted with thanks and i will study the code in detail on how to save the files.

In regarding to the intrinsic matrix, any example of using it to creat point clouds?  I will use open3D as the tool.

marek-simonik, 2021-08-09:
In [this C++ library](https://github.com/marek-simonik/record3d_unity_streaming/blob/f9d5cdb33db815dc55e77cb12e2c740d89432b7e/Record3DUnityStreaming.cpp#L111), you can see how I generate point cloud out of the RGBD data.

lamhoson, 2021-08-09:
Thank you very much.
I am using Python for this project. Any Python function API interface I could call to this C++ function?

marek-simonik, 2021-08-09:
No problem. I think that instead of using the C++ library, it might be better if you implemented something similar using `numpy`. Or you can take a look at [this project](https://github.com/kentaroy47/ipad-lidar-stream), which likely does what you need.

lamhoson, 2021-08-09:
Thanks!

取得 iOS 版 Outlook<https://aka.ms/o0ukef>
________________________________
寄件者: Marek Šimoník ***@***.***>
寄件日期: Monday, August 9, 2021 10:32:06 PM
收件者: marek-simonik/record3d ***@***.***>
副本: Big Horse ***@***.***>; Author ***@***.***>
主旨: Re: [marek-simonik/record3d] usbmuxd_get_device_list: error opening socket! (#21)

No problem. I think that instead of using the C++ library, it might be better if you implemented something similar using numpy. Or you can take a look at this project<https://github.com/kentaroy47/ipad-lidar-stream>, which likely does what you need.

lamhoson, 2021-08-10:
Any documentation about the captured depth map data format, say 8-bit, floating point or else etc?

I just see pure dark image from the depth map jpg file capture.

marek-simonik, 2021-08-10:
The depth frames received via USB streaming are 32bit `float`s. They are in the range of approx. 0-3 meters, which explains why you see black image when you save it into a JPEG file (the pixel values will be mostly 0-3 out of 255 for an 8bit JPEG). You will need to normalize the depth image if you want to see a grayscale visualization.

lamhoson, 2021-08-10:
Noted with thanks.  Then how about the RGB image, same 32bits float?

取得 iOS 版 Outlook<https://aka.ms/o0ukef>
________________________________
寄件者: Marek Šimoník ***@***.***>
寄件日期: Tuesday, August 10, 2021 6:21:51 PM
收件者: marek-simonik/record3d ***@***.***>
副本: Big Horse ***@***.***>; Author ***@***.***>
主旨: Re: [marek-simonik/record3d] usbmuxd_get_device_list: error opening socket! (#21)

The depth frames received via USB streaming are 32bit floats. They are in the range of approx. 0-3 meters, which explains why you see black image when you save it into a JPEG file (the pixel values will be mostly 0-3 out of 255 for an 8bit JPEG). You will need to normalize the depth image if you want to see a grayscale visualization.

marek-simonik, 2021-08-10:
The RGB image is ordinary 3-channel image (R, G, B). Each channel is 8bit `uint8` (so together 3x8 = 24 bits).

Both the depth map and the RGB image are `numpy` arrays, so you can see their resolution and number of channels by printing `image.shape`. To see their datatype, print `image.dtype`.