marek-simonik/record3d #25: What dose High quality RGBD stream mean?

jackchinor, 2021-11-12:
Hi, thanks for sharing this repo, What dose High quality RGBD stream mean? High quality RGBD means higher resolution or something else? Can I get high resolution(640*480) RGB and 256*192 deph map ? hoping for your kind reply. Thanks.

marek-simonik, 2021-11-12:
Hi,
did you mean the "*Higher quality LiDAR recording*" option in the Record3D Settings section? If you enable this option, you will get 720x960 px RGB images and 192x256 px depth images for LiDAR. If you disable the option, then both the RGB and the depth images will be 192x256 px.

I think this might answer your question, so I am closing this issue, but feel free to add a new comment if something would not be clear (or send me an email).

closed 2021-11-12 by marek-simonik

jackchinor, 2021-11-12:
> Hi, did you mean the "_Higher quality LiDAR recording_" option in the Record3D Settings section? If you enable this option, you will get 720x960 px RGB images and 192x256 px depth images for LiDAR. [...]

Thank you very much. That's what I want.