marek-simonik/record3d #29: CMAKE - errors

fearboo, 2022-01-20:
I must be missing some prerequisites for USB setup.
I can't figure it out.  I have installed VS 2022  and 2019 and have CMake tools installed.

I just keep getting errors for Cmake.

I installed it to path for all users.

Any help would be great.

I have tried using the Wifi method but its doesnt seem to work in Brave or Edge. I am not installing chrome.

Thanks.

marek-simonik, 2022-01-20:
Hello,

can you please specify which errors do you get (either a screenshot or the text of the errors)?

Regarding Wi-Fi streaming: as far as I know, Chromium-based browsers changed their privacy defaults, so to be able to use the Wi-Fi streaming demos in a Chromium-based browser (e.g. Chrome, Brave, Edge), please follow [this SO answer](https://stackoverflow.com/a/69464875) and disable `Block insecure private network requests` in Brave's flags as is suggested in the SO answer. Then my Github Wi-Fi streaming demos should work properly.

(Note that mobile Safari doesn't suffer from this problem (at least not on iOS 15), so the Wi-Fi streaming demo works without issues in iOS/iPadOS Safari.)

fearboo, 2022-01-23:
I am using windows 10, 64bit on an Intel.  I notice it's using a build library for win-amd64 

(base) PS I:\code\record3d> python setup.py install
running install
running bdist_egg
running egg_info
writing record3d.egg-info\PKG-INFO
writing dependency_links to record3d.egg-info\dependency_links.txt
writing requirements to record3d.egg-info\requires.txt
writing top-level names to record3d.egg-info\top_level.txt
reading manifest file 'record3d.egg-info\SOURCES.txt'
reading manifest template 'MANIFEST.in'
adding license file 'LICENSE'
writing manifest file 'record3d.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_ext
CMake Error at CMakeLists.txt:2 (project):
  Generator

    NMake Makefiles

  does not support platform specification, but platform

    x64

  was specified.

CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
See also "I:/CODE/record3d/build/temp.win-amd64-3.9/Release/CMakeFiles/CMakeOutput.log".
Traceback (most recent call last):
  File "I:\code\record3d\setup.py", line 68, in <module>
    setup(
  File "G:\studio_software\Anaconda\lib\site-packages\setuptools\__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "G:\studio_software\Anaconda\lib\distutils\core.py", line 148, in setup
    dist.run_commands()
  File "G:\studio_software\Anaconda\lib\distutils\dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "G:\studio_software\Anaconda\lib\distutils\dist.py", line 985, in run_command
    cmd_obj.run()
  File "G:\studio_software\Anaconda\lib\site-packages\setuptools\command\install.py", line 67, in run
    self.do_egg_install()
  File "G:\studio_software\Anaconda\lib\site-packages\setuptools\command\install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "G:\studio_software\Anaconda\lib\distutils\cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "G:\studio_software\Anaconda\lib\distutils\dist.py", line 985, in run_command
    cmd_obj.run()
  File "G:\studio_software\Anaconda\lib\site-packages\setuptools\command\bdist_egg.py", line 164, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "G:\studio_software\Anaconda\lib\site-packages\setuptools\command\bdist_egg.py", line 150, in call_command
    self.run_command(cmdname)
  File "G:\studio_software\Anaconda\lib\distutils\cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "G:\studio_software\Anaconda\lib\distutils\dist.py", line 985, in run_command
    cmd_obj.run()
  File "G:\studio_software\Anaconda\lib\site-packages\setuptools\command\install_lib.py", line 11, in run
    self.build()
  File "G:\studio_software\Anaconda\lib\distutils\command\install_lib.py", line 107, in build
    self.run_command('build_ext')
  File "G:\studio_software\Anaconda\lib\distutils\cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "G:\studio_software\Anaconda\lib\distutils\dist.py", line 985, in run_command
    cmd_obj.run()
  File "I:\code\record3d\setup.py", line 34, in run
    self.build_extension(ext)
  File "I:\code\record3d\setup.py", line 59, in build_extension
    subprocess.check_call(['cmake', ext.sourcedir] + cmake_args, cwd=self.build_temp, env=env)
  File "G:\studio_software\Anaconda\lib\subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', 'I:\\code\\record3d', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=I:\\code\\record3d\\build\\lib.win-amd64-3.9', '-DPYTHON_EXECUTABLE=G:\\studio_software\\Anaconda\\python.exe', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY_RELEASE=I:\\code\\record3d\\build\\lib.win-amd64-3.9', '-A', 'x64', '-DBUILD_PYTHON_BINDINGS=ON']' returned non-zero exit status 1.

marek-simonik, 2022-01-23:
The `win-amd64` is Microsoft's way of describing a 64bit platform [due to historical reasons](https://answers.microsoft.com/en-us/windows/forum/all/why-are-there-amd64-files-on-an-intel-machine/7482ee8f-511a-45c0-b3a3-8542a74801c9) — it's the same for both Intel and AMD CPUs, so that is not an issue.

The most interesting part of the log you shared is:
```
Generator NMake Makefiles does not support platform specification, but platform x64 was specified.
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
```

This suggests you might not have Visual Studio installed on your computer. Please make sure there is a Visual Studio installed on your computer, try again and let me know if that helped. If you do have a VS installed and you still see the exact same error, then CMake for some reason cannot see VS.

fearboo, 2022-01-23:
This is a problem on my end somehow. I just can't figure it out.  I am using Anaconda . I have MS VS 2022 and 2019 installed.    I have tried installing Cmake install file to system path.    I have tried installing cmake through MS VS installer. I have tried Cmake install package from the anaconda Navigator.             I have restarted my machine.

I checked my windows ENV settings.  I am not sure exactly what paths I should have for MS VS to work.

I had an issue like this before and could never solve it.
I have tried reinstalling MS VS many times.

Any help you could give would be great.
I really don't want to reinstall windows, but I have no idea why python can't see my VS.

marek-simonik, 2022-01-24:
I don't think the issue is related to Python per se; I think that even if you tried to build the `record3d` C++ demo project using e.g. `ccmake`, then the GUI still wouldn't provide you a MSVC compiler as an option. But you can try that though and maybe even select the path to the compiler manually; try running `ccmake` (should be installed alongside Cmake) and attempt to build the C++ demo.

I found [this Stack Overflow question](https://stackoverflow.com/questions/31619296/cmake-does-not-find-visual-c-compiler), which seems to be about the same issue you are facing. I suggest that you see if any of the answers will help.

marek-simonik, 2022-11-08:
I'm closing this issue due to inactivity, but feel free to comment and I'll try to help.

closed 2022-11-08 by marek-simonik