marek-simonik/record3d #30: Several questions on updated resolution

zehuiz2, 2022-01-28:
Hi,
I wonder if you've updated both LiDAR & FaceID depth resolution? 

`depth_img = depth_img.reshape((1280, 960))  # For a FaceID camera 3D Video`
 `depth_img = depth_img.reshape((512, 384))  # For a LiDAR 3D Video`

Is the above correct? 
Another three questions:

1. Is it possible you could update the FaceID RGB resolution? It is still 640*480.
2. It seems the LiDAR confidence resolution is not updated?
3. What are the units of the depth measurements? I assume it is mm?

Thank you very much!

_Originally posted by @zehuiz2 in https://github.com/marek-simonik/record3d/issues/7#issuecomment-1024665589_

marek-simonik, 2022-01-29:
Hi,
I have not updated resolution of any output — all resolutions stay the same (as far as I know, Apple has not increased depth resolutions in iOS 15, so 192x256 px for LiDAR and 480x640 px for FaceID are the maximums, therefore I do not plan any changes yet). The resolution of LiDAR confidence maps should also be 192x256 px. To answer your third question: the depth measurements are in meters.

zehuiz2, 2022-01-29:
Thank you for the reply. 

However, using 192*256 for LiDAR and 480*640 for FaceID, I got the following errors:
`cannot reshape array of size 1228800 into shape (640,380)`
`cannot reshape array of size 196608 into shape (256,192)`

As for the units, it does not make sense to me the depth is in meters. For example, with the LiDAR data, I got the following
`array([[  0,  96, 184, ..., 160, 252,  62],
       [  0,  96, 254, ...,   0,  97,  63],
       [  0,   0, 184, ...,   0, 254,  62],
       ...,
       [  0,  32,  11, ..., 224, 116,  63],
       [  0,  32, 194, ...,   0,  11,  63],
       [  0,  32,  11, ...,  64, 124,  63]], dtype=uint8)`
I took images at around 0.4m. Thank you.

closed 2022-01-29 by zehuiz2

marek-simonik, 2022-01-30:
> Thank you for the reply. I figured it out:
> I mistakenly used:
> `depth_img = np.frombuffer(decompressed_bytes, dtype=np.uint8)`

That is not the correct way of loading the depth image. Please see this sample script: https://github.com/marek-simonik/record3d/issues/7#issuecomment-736573410

The issue with the snipped you posted is that you interpret the bytes as `uint8`. You should instead load depth maps as `float32` — then you will get depth values in meters.

Only the confidence maps are to be loaded as `uint8`.