marek-simonik/record3d #31: Few Questions Intrinsic and Extrinsic Parameters

zehuiz2, 2022-01-29:
First, I must say this is a great App and I would like to thank the developer for being responsive!
I have some questions regarding the intrinsic and extrinsic parameters:

1. Is the depth image and RGB image already fused? Do I need to estimate the rigid transformation between the lidar and the camera?
2. Do I need to calibrate the RGB/Depth images in terms of radial and tangential distortions?
3. I have difficulty understanding the `poses` matrix provided in `metadata`. From previous posts, I think there are extrinsic parameters. I assume each row corresponds to each image. There are seven parameters in each row. How could extrinsic parameters have only 7 elements? I guess it should have 12?

I'm looking forward to hearing from you!

areiner222, 2022-01-29:
I am another enthusiastic user of this app - working great for me!

@zehuiz2 on question 3: extrinsic parameters - take a look at [this issue](https://github.com/marek-simonik/record3d/issues/27). Also, I believe each row corresponds to each image.

zehuiz2, 2022-01-29:
> I am another enthusiastic user of this app - working great for me!
> 
> @zehuiz2 on question 3: extrinsic parameters - take a look at [this issue](https://github.com/marek-simonik/record3d/issues/27). Also, I believe each row corresponds to each image.

Thank you! It really helped. One question as a beginner, could I derive the perspective transformation matrix given two poses of the camera?

zehuiz2, 2022-01-29:
One follow-up question on question 3. Given the quaternion, I could restore the $$M_11$$

marek-simonik, 2022-01-30:
I will first answer your original questions:

1. It should be fused internally by Apple.
2. Similar to 1., Apple should have already taken care of distortion correction.
3. As @areiner222 correctly answered with the link, the poses are stored as a quaternion + world pose.

To answer your follow-up questions:
1. The quaternion is stored as `qx, qy, qz, qw`.
2. Yes, exactly.
3. I think it should be `X_{world} = [R|t] K X_{cam}`.

closed 2022-01-31 by zehuiz2