marek-simonik/record3d #34: How to get higher-resolution RGB images (1920 x 1440)?

andrewsonga, 2022-03-11:
Hello,

first of all, thank you so much for your work. It has truly made extracting RGBD video from iphone / ipads a lot easier! I just wanted to ask whether it's possible to extract higher-resolution RGB images than the current max. resolution that is outputted by the app (960 x 720). I've been playing around with ARKit and it seems like the resolution of the capturedImage (RGB) is (1920 x 1440); is it possible to capture RGB video at this resolution (1920 x 1440) on record3d as well?

Thank you in advance :)

marek-simonik, 2022-03-13:
Hello,

thank you for using Record3D :)! I had to lower the resolution to 720×960 px due to performance issues, but 1440×1920 px RGB images might come in a future update if I manage to improve the performance.

P.S. I might reply faster via email (support@record3d.app).

marek-simonik, 2022-11-08:
The option to record 1440x1920 px RGB images for LiDAR has been added in Record3D 1.8. To enable this option, navigate to the Settings tab and turn on "Higher-quality LiDAR recording".

closed 2022-11-08 by marek-simonik