marek-simonik/record3d #38: Frames never arrive

carsonswope, 2022-04-19:
Hi, I'm attempting to stream the depth frames to my windows PC. I installed itunes so that the USB interface works. When I attempt to run demo.exe, I get this in the log:

`Found 1 iOS device(s):
        Device ID: 4776
        UDID: 00008101-000D392C01F0001E

Trying to connect to device with ID 4776.
Connected and starting to stream. Enable USB streaming in the Record3D iOS app (https://record3d.app/) in case you don't see RGBD stream.`

And then it just hangs there. When I stop the frame from the record3D app, it says 'Stream Stopped!' but the program never exits.

I've traced the problem to the `usbmuxd_recv` call on Record3DStream.cpp:243 called when attempting to load the PeerTalk header. It seems that it's not receiving any bytes: numTotalReceivedBytes stays at 0. 

Any ideas? :) thanks

carsonswope, 2022-04-19:
Nevermind... something was messed up with the truedepth camera. Restarting the iphone fixed it.

closed 2022-04-19 by carsonswope

marek-simonik, 2022-04-19:
Hi,

I'm glad to hear the stream started to work after a restart :)!

If the stream won't start again in the future, I would recommend to first try force-quitting just Record3D and if that wouldn't help, only then proceed with full device restart (but I guess you already tried that).