marek-simonik/record3d #40: C++ demo app not working on Linux

jaykim305, 2022-08-08:
Hi, thank you for your awesome project.

I have the exact same issue as [this](https://github.com/marek-simonik/record3d/issues/38) but on Linux Ubuntu. 
I checked that the python demo-main.py works just fine on the same machine.
But when I run the ./demo after a successful build, it just hangs and OpenCV display is not shown as supposed to. 
Below is the config result after I run `cmake -DCMAKE_BUILD_TYPE=Release`.

```
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test CFLAG_Wall
-- Performing Test CFLAG_Wall - Success
-- Performing Test CFLAG_Wunknown_pragmas
-- Performing Test CFLAG_Wunknown_pragmas - Success
-- Performing Test CFLAG_Wunused_variable
-- Performing Test CFLAG_Wunused_variable - Success
-- Looking for stpncpy
-- Looking for stpncpy - found
-- Looking for sleep
-- Looking for sleep - not found
-- Looking for pselect
-- Looking for pselect - found
-- Looking for sys/inotify.h
-- Looking for sys/inotify.h - found
-- Found OpenCV: /usr (found version "3.2.0")
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jaykim305/record3d/build
```

Any ideas?

jaykim305, 2022-08-08:
I think there exists a deadlock here which causes the problem. https://github.com/marek-simonik/record3d/blob/7a3764ba0e516e7905dafe9f413dd8f13955f4ce/src/DemoMain.cpp#L71 If I put `usleep(1)` before `continue`, now the program works in a loop without freeze. But this is just a half-measure for now I guess.

Also, just a minor thing: perhaps `unistd.h` instead of `string.h` in https://github.com/marek-simonik/record3d/blob/7a3764ba0e516e7905dafe9f413dd8f13955f4ce/libs/unix/libusbmuxd/CMakeLists.txt#L12

marek-simonik, 2022-08-12:
Thank you for reporting this bug, I confirm that I can replicate it. It will be fixed in the upcoming version of this C++ library, which will be available with the release of Record3D 1.7.2 by the end of the next week.

marek-simonik, 2022-08-16:
Please update Record3D to version 1.7.2, update this library to version 1.3.1 and let me know if you still experience the issue with deadlock. Thanks.

jaykim305, 2022-08-17:
Thank you! I will check out the latest version.

closed 2022-08-17 by jaykim305