marek-simonik/record3d #45: Extending the record depth

hrwu0, 2022-09-20:
Hi!

I would like to know if the maximum of depth could be expanded, I believe it’s set to be around 3 meters as default, but the LiDAR of iPhone supports 5 meters. Is there any way for us to manually change the maximum depth?

Also, I wonder if the new API coming with iOS15.4 is supported.

Thank you!

marek-simonik, 2022-09-20:
Hi,

the next update of Record3D will remove the maximum depth restriction from recorded videos. The update will be released either by the end of this week or early in the next week.

I will also attempt to add the option to optionally enable the new iOS 15.4 LiDAR API — but note that when using this variant, there will be no tracking (i.e. no pose metadata).

marek-simonik, 2022-09-28:
I don't think it would be possible to weaken the requirement of iOS/iPadOS 16 without a performance hit; iOS 16 brought new APIs that made it possible for me to implement 2x LiDAR RGB resolution and other under-the-hood performance improvements. Supporting iOS 15 would mean removing an existing feature from Record3D (a feature which other users requested).

I think that iPadOS should be released in October (I've been testing Record3D on iPadOS 16 Beta) and the public version of iOS 16 has already been released.

andybak, 2022-11-08:
Another thought - use a logarithmic mapping or some other non-linear mapping?

jingyibo123, 2022-11-16:
> I don't think it would be possible to weaken the requirement of iOS/iPadOS 16 without a performance hit; iOS 16 brought new APIs that made it possible for me to implement 2x LiDAR RGB resolution and [...]

I'm not a IOS developper, but is there a way to support iOS15&16 simutaneously?
Updating older device (iPhone 12 pm) to latest iOS is not always an option....

Otherwise, is it possible to maintain a iOS15 compatible branch? Or maybe just point to the last compatible commit here so I could fork a branch.

BTW. Thanks for the great work!

marek-simonik, 2022-11-16:
Thanks for using Record3D, @jingyibo123 :)!

Let me separate the Record3D iOS app and (this) `record3d` library:

### The Record3D iOS app

Unfortunately, Apple does not allow developers to simultaneously offer different versions of iOS apps on the App Store — only the latest update is available for download on the App Store, but there might be a workaround.

The iOS 16 requirement was introduced in Record3D 1.8. If you cannot or do not want to update your iPhone to iOS 16, then App Store might let you download an older version of Record3D if you have already used your Apple ID to download Record3D in the past; in that case, just open [Record3D's App Store page](https://apps.apple.com/us/app/record3d-3d-videos/id1477716895?ls=1) and tap the download button (App Store will ask you if you want to download and older version of Record3D — [this website shows a screenshot of how that prompt looks like](https://www.imobie.com/support/how-to-install-old-version-of-app.htm)).

In case you haven't downloaded Record3D yet and if you, at the same time, have multiple iPhones/iPads that you are logged into with the same Apple ID, and if one of the devices is running iOS 16, then you can download the latest Record3D version onto the iOS 16 iDevice, which should make it possible for you to download an older version of Record3D your iPhone 12 Pro Max running iOS 15.

(Please note that all iPhones/iPads capable of recording 3D videos can be updated to iOS 16 ([compatible devices](https://support.apple.com/en-us/HT213411)), but I can understand that you might prefer to stay away from iOS 16.)

### The `record3d` library

The compatibility of the `record3d` library is tied to the version of the Record3D iOS app (i.e. not to the version of iOS per se).

You can use the latest version of the `record3d` library (i.e. branch [`version/1.3.1`](https://github.com/marek-simonik/record3d/tree/version/v1.3.1)) with Record3D 1.7.2 or newer. Record3D 1.7.2 was compatible with iOS 14.1+, so that makes it compatible with iOS 15 too.

If you use an older Record3D version, then you should download an appropriate version of the `record3d` library under the [Releases section](https://github.com/marek-simonik/record3d/releases).

jingyibo123, 2022-11-17:
That's a great trick to keep old device compatible of new apps! 

Thank you Marek, keep up the great work~