marek-simonik/record3d #46: matrix data stream not visible

mminikk, 2022-09-29:
I am a student and I need to memorize in a txt file the straming RGBD from an iPhone connected to a robot. On the robot there is no montor. I installed the C++ version of record3d and demo app on the robot (operating system of the robot is Ubuntu 18.04 and ROS melodic).
If I log by ssh to the robot (using flag -X) the demo app seems works, I see the two windows with optical and lidar data images but on the terminal window I cannot see the textual matrix of data that is what I need.
Can you help me? I'm not a programmer but a electronic student.
Thank you.

marek-simonik, 2022-09-29:
Hello,
the purpose of the C++ and Python demos is just to demonstrate how you can use the `record3d` library; it's not meant as a full-fledged app. I meant it just as a sample code, which people can tweak for their specific needs, so it is expected that there is no textual output in the C++ demo. 

I am not sure if I correctly understood what you wish to do and what you meant by "*the textual matrix of data*". If I am not mistaken, then you wish to print the value of each RGB + Depth pixel to the program output (`stdout`)? Even though this is possible to implement, I don't think you want to do this as you would get terrible performance (the live stream would start to lag).

I would recommend to ask someone who has experience with ROS how you could efficiently save the live data stream. My ROS experience is very limited, so I am unfortunately not qualified to help. I recommend that you ask e.g. on StackOverflow.

To at least partially answer your question and provide some starting point: if you wish to process the incoming frame data (RGB and Depth images), you can add new code here: https://github.com/marek-simonik/record3d/blob/d712f4026d26a6cf3c7f697e4199ae5bf47395f1/src/DemoMain.cpp#L123

mminikk, 2022-09-30:
Thanks Marek, you got it right, I would like to save the program output (stdout) and then process the data. You were very kind to tell me where to add the code to process the incoming frame. Performance isn't critical to me, I now work at 6 FPS. Thanks again for your help.