marek-simonik/record3d #48: Getting an error ... Iphone 12 Pro plugged into a windows 10 machine

gschian0, 2022-11-27:
Is is possible to plug in the iphone 12 to windows 10 and use the python app?

`Searching for devices
[libusbmuxd] usbmuxd_get_device_list: error opening socket!
0 device(s) found
Traceback (most recent call last):
  File "C:\Users\GennaroSchiano\Documents\GitHub\record3d\demo-main.py", line 76, in <module>
    app.connect_to_device(dev_idx=0)
  File "C:\Users\GennaroSchiano\Documents\GitHub\record3d\demo-main.py", line 31, in connect_to_device
    raise RuntimeError('Cannot connect to device #{}, try different index.'
RuntimeError: Cannot connect to device #0, try different index.`

marek-simonik, 2022-11-27:
Yes, it should be possible. The error you are getting suggests that Windows doesn't "see" your iPhone (this has nothing to do with the `record3d` library).

Do you have iTunes installed on Windows and can you see the details of your iPhone in iTunes? Having iTunes installed is essential for USB streaming to work. To be able to stream via USB, you also need to press the *Trust* button when the [*Trust This Computer?* pop-up](https://support.apple.com/en-us/HT202778) shows up on your iPhone.

marek-simonik, 2023-02-01:
It's been some time since this issue was active. I assume you were able to get the Python demo running eventually. Feel free to reopen this issue if you still experience this problem.

closed 2023-02-01 by marek-simonik