marek-simonik/record3d #50: Segmentation fault (core dumped)

kaimingkuang, 2022-12-14:
I am trying to stream RGBD video using record3d but I run into segmentation fault every single time:
```
1 device(s) found
        ID: 4776
        UDID: 00008101-0011348121B8001E

Camera Intrinsics: [[435.33657837   0.         240.14639282]
 [  0.         435.33657837 320.78341675]
 [  0.           0.           1.        ]]
INFO: Created TensorFlow Lite XNNPACK delegate for CPU.
Segmentation fault (core dumped)
```
These are the softwares/hardwares I am using:
- iPhone 12 pro max;
- iOS 16.2;
- Record3D app 1.8.3;
- record3d library 1.3.0 via pip;
- Ubuntu 20.04.4 LTS 64 bit;
- AMD Ryzen pro 3975wx CPU with NVIDIA RTX A4000.

Thank you!

marek-simonik, 2022-12-14:
Please update the `record3d` library to the latest version (1.3.1) — either via `pip` or by installing from source. The Record3D iOS app version 1.7.2 and newer has to be used with the `record3d` library version 1.3.1 (or newer).

marek-simonik, 2023-02-01:
I assume the segfault has been fixed by updating to the latest version of Record3D iOS app and library so I'm closing this issue. Feel free to reopen if the problems persist.

closed 2023-02-01 by marek-simonik