marek-simonik/record3d #52: Setting JPEG compression for in-app captures

swtyree, 2023-01-26:
Hi @marek-simonik, thanks for the fantastic app! I use depth cameras for computer vision research, and the ability to capture RGB-D videos without being tethered to a computer is a game-changer.

My one complaint is that the JPEG compression rate for in-app captures leaves a lot of artifacts. Are there any plans to expose that setting in the future? I've made a few captures using the USB interface with lossless RGB streaming, and the resulting images are much better. I realize that a higher setting will significantly increase file sizes on the iPad, but that is something I am willing to work around.

Thanks!

marek-simonik, 2023-01-29:
Hi @swtyree and thanks for using Record3D :).

Apart from increased file size (which isn't necessarily a problem, as you mentioned), there is the issue of ensuring real-time performance when processing (nearly) lossless images. I'll take a look into this to see how performance would be affected across different devices under low levels of compression and if it turns out there is a negligible performance drop, I will try to add such option into a future update.

P.S. The lossless USB streaming (`USB Streaming RGB quality` exactly at `100`) uses PNG instead of JPEG, so if your use case requires lossless quality, then I don't think it would be possible to implement this feature in real-time (PNG encoding isn't hardware-accelerated).

swtyree, 2023-01-30:
Thanks, @marek-simonik! I think a range of JPEG compression settings would be acceptable. For instance, I tried settings from 70-100 for a frame from a recent lossless capture (`cv2.imwrite(filename, img, [cv2.IMWRITE_JPEG_QUALITY, quality])`). Beyond 75 or 80, the visual differences were barely perceptible, while the file sizes were noticeable smaller:

```
quality_070.jpg  28K
quality_075.jpg  31K
quality_080.jpg  36K
quality_085.jpg  43K
quality_090.jpg  56K
quality_095.jpg  86K
quality_100.jpg 188K
quality____.png 428K
```

marek-simonik, 2023-01-31:
OK, if (limited range of) JPEG compression settings is an acceptable compromise for you, then I will add such option into the next Record3D update.

swtyree, 2023-01-31:
Thank you! Do you have an estimate for when the next update will be pushed?

marek-simonik, 2023-02-01:
The next Record3D update should be on the App Store next week (or earlier).

closed 2023-02-01 by swtyree