marek-simonik/record3d #54: About depth map

HquGhy, 2023-02-01:
Whether the python file in the demo needs to be modified in order to obtain the depth map？

marek-simonik, 2023-02-01:
I'm not sure if I understood your question. The Python demo does already give access to the depth map: https://github.com/marek-simonik/record3d/blob/d8637f57702faa6224c27550ab2b689791f9ac50/demo-main.py#L50

HquGhy, 2023-02-01:
I mean, I want a depth map in png format

marek-simonik, 2023-02-01:
If you want to save the depth map [provided in the Python demo](https://github.com/marek-simonik/record3d/blob/d8637f57702faa6224c27550ab2b689791f9ac50/demo-main.py#L50) into PNG format, then you will have to do it yourself by modifying the code according to your needs, because the Python demo does not have such capability (the demo is meant just to illustrate how to use the API).

closed 2023-02-01 by marek-simonik