marek-simonik/record3d #57: Customize Colormaps on USB Live Streaming Mode

LaThanhTrong, 2023-03-06:
When I am running `demo-main.py`, it shows only black and white background. I tried to switch colormaps in the settings, it does not work. It only works when streaming mode is turned off. Is there anything I have to do in the settings or lines of code can solve this issue?

LaThanhTrong, 2023-03-08:
Do python also support shading aswell?

LaThanhTrong, 2023-03-08:
> Do you mean that you see the depth map in shades of gray instead of colored by the currently selected colormap from the Settings section? It is not a bug, but the expected behavior; the USB streaming mode is supposed to stream unaltered RGBD data and it is the responsibility of the receiver (e.g. the Python script) to post-process the raw output into the desired output.
> 
> I would advise to adjust the Python script and apply color mapping (or other post-processing you need) to the depth map via Python. Note that the depth map contains `float32` numbers and invalid values are stored as `NaN`s (something to be aware of when implementing color mapping).

Do python also support shading aswell?

marek-simonik, 2023-03-10:
You can implement arbitrary post-processing in Python, but there is no out-of-the-box support for shading per se; you will have to implement your own filters/post-processing from scratch (it is unrelated to Record3D).

pcktm, 2023-03-31:
An example of color mapping:
```python
self.max_depth_ever = max(self.max_depth_ever, np.nanmax(depth))
depth[np.isnan(depth)] = self.max_depth_ever
depth = depth / self.max_depth_ever * 255
depth = depth.astype(np.uint8)

depth = cv2.applyColorMap(depth, cv2.COLORMAP_JET)
```