marek-simonik/record3d #61: [feature request]  Remove auto-focus

RaiAnant, 2023-04-10:
Hi there!

I've been using the Record3D app for my research and it has been incredibly helpful so far. However, there is one thing I am noticing when I am recording videos using the app, sometimes it focuses on the wrong things (or things I don't care about). This ends up blurring a lot of other details that are important.

I've searched for a way to manually set focus or adjust the app's behaviour, but I couldn't find any such feature. Therefore, I'd like to request the addition of manual focus control to the app's features. I believe that this feature would be valuable not only for us but also for other users who need to control the focus of their recordings.

Thank you!

marek-simonik, 2023-04-12:
Hi, and thank you for using Record3D!

I think it is unfortunately not possible to implement full manual focus support yet (at least not with the current public version of iOS and on the iDevices I own). I am only aware of the ability to optionally disable autofocus, which is already implemented in Record3D. This option is available in the `Settings tab > Export options > Camera autofocus`.

I understand this is not a solution to the problem you described, but it might at least help you lock focus.

RaiAnant, 2023-04-19:
Thanks a lot! Disabling autofocus will be very helpful. Also on a side note, is there a way to have an RGB viewfinder inside record3d? While recording, the viewfinder only shows a gray depth map/ surface map. Is there a way to switch to a traditional RGB viewfinder?