marek-simonik/record3d #62: Exporting/reading saved sequences

sandeepnmenon, 2023-04-10:
I have been testing out record3d for the datacollection at my lab and want to build some datapipelines around it. 
With this library I see that the API is mainly to deal with the streaming data.
Is there an API to read captured sequences from the phone or export them via api to my code?

PS: I would like to bypass the manual process of exporting them from the app through the UI

andybak, 2023-04-10:
Oh yes! iOS is such a horror show for moving data between different apps or workflows. Anything that makes this easier would be great.

marek-simonik, 2023-04-12:
There is no such API available today, but thank you for the suggestion. I do not think it would be easy to add the ability to pass convert videos into USB stream instead of exporting them to the disk, so the export process would likely still need to be done inside the app.

However, copying already exported data is a more realistic scenario (although I would need to ensure privacy by letting the user of the iOS app activate a "data-transfer mode"). The downside of this approach is that you would need to keep Record3D open during the whole time your data would be copied (which can take a while given that most iDevices still run at USB 2.0 speeds).

An alternative way to simplify the process of copying files from an iDevice via USB cable (without needing to have Record3D permanently open on screen) would be to use a library like [libimobiledevice](https://libimobiledevice.org) to access File Sharing capabilities, which would allow you to access the file system of your iDevice (beyond only Record3D).