marek-simonik/record3d #64: Outdoor Environment Camera Pose

tb2-sy, 2023-05-19:
Is it reasonable to use record3d to obtain the camera pose in an outdoor environment? Because there are no cases of outdoor environments in many examples.

marek-simonik, 2023-05-19:
The LiDAR poses provided by Record3D are (currently) estimated by ARKit, which seems to work reasonably well in both indoor and outdoor environment. There is no particular reason why I have not yet shown a sample outdoor LiDAR video.

TimYao18, 2023-12-15:
Are the poses converted into OpenGL coordinate system from ARKit `arframe.camera.transform` when recording videos?

marek-simonik, 2023-12-16:
> Are the poses converted into OpenGL coordinate system from ARKit `arframe.camera.transform` when recording videos?

Yes, they are; the poses reported by Record3D use the OpenGL coordinate system convention.

marek-simonik, 2024-06-09:
> Thanks for the great app, with ARKit do you mean the apple's ARKit https://developer.apple.com/documentation/arkit/arframe ?

Thank you for using Record3D. Yes, I do — exactly as you wrote.