marek-simonik/record3d #68: demo-main.py - ModuleNotFoundError: No module named 'cv2'

benlegentil, 2023-06-06:
Hello,

I've made it to the 'Sample Applications' part of the ReadMe (I think). When I try to run "python demo-main.py" from window's command I get:

Traceback (most recent call last):
  File "C:\...\demo-main.py", line 3, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'

_It's probably extremely easy for anyone familiar with coding but a quick recorded video would go a long way to know what's the expected behaviour of having everything runs properly._

marek-simonik, 2023-06-06:
Hello,

the error says that you do not have the `cv2` Python module installed. The `cv2` module is OpenCV. To fix this issue, please install OpenCV via pip by running this in the Command Line: `python -m pip install --user opencv-python`

The expected behavior of `demo-main.py` is that 2 windows show up, one window displaying the live RGB camera stream and the second window displaying the live depth map stream.

Let me know if you'd run into further problems.

benlegentil, 2023-06-07:
Thank you! Solved it right away and I was able to see the two windows

closed 2023-06-07 by benlegentil