marek-simonik/record3d #69: building for m1

gschian0, 2023-06-13:
I get this error ... 
" for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: warning: ignoring file /opt/homebrew/lib/libopencv_photo.4.7.0.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: warning: ignoring file /opt/homebrew/lib/libopencv_core.4.7.0.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: warning: ignoring file /opt/homebrew/lib/libopencv_imgproc.4.7.0.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64"

how do I switch the version to build for arm64?

thanks for your time this program is awesome :)

gschian0, 2023-06-13:
it was just a warning... I reinstalled opencv and I am good.

dodgeroo, 2024-04-30:
I am trying to open on mac M1 and I get this error:
line 1, in <module>
    import numpy as np
ModuleNotFoundError: No module named 'numpy'

I can see the 'numpy.h' file in the pybind11 folder.

not sure what I am doing wrong?

marek-simonik, 2024-04-30:
> I am trying to open on mac M1 and I get this error: line 1, in import numpy as np ModuleNotFoundError: No module named 'numpy'
> 
> I can see the 'numpy.h' file in the pybind11 folder.
> 
> not sure what I am doing wrong?

The error sounds like your Python installation doesn't have `numpy` installed. Try to install `numpy` via `pip` (that should fix the error):

```
python3 -m pip install numpy
```

dodgeroo, 2024-05-07:
Hi,
I got a bit futher but now don't know how to get CMake into the "PATH". Which path is it referring to?
Best,

dodgeroo, 2024-05-07:
now it says ModuleNotFoundError: No module named 'record3d'

i don't know what I'm doing wrong here. i seem to be doing everything correctly

i can only get it working on Safari browser. not usb python script.

marek-simonik, 2024-05-11:
I apologize for not replying earlier.

> I got a bit futher but now don't know how to get CMake into the "PATH". Which path is it referring to?

"PATH" refers to the `PATH` environment variable. If you install CMake via Homebew, you won't have to adjust `PATH`.

> i can only get it working on Safari browser. not usb python script.

I assume you meant that Wi-Fi streaming does work correctly for you, but the USB streaming does not?

> now it says ModuleNotFoundError: No module named 'record3d'

This error means that the `record3d` Python package has not been installed yet. You can build it from source, but in order to do that, you will need to install Xcode from the App Store (although I think that installing only Xcode Command Line Tools should be enough). Once you will have that installed, open Terminal and execute the following:

```bash
cd /tmp
git clone https://github.com/marek-simonik/record3d.git
cd record3d
python3 -m pip install .  --user
```

If the installation was successful, you should see "*Successfully installed record3d-1.3.1*".

Let me know if there will be any issues.

ofergetz, 2025-04-17:
Hello!
I get this error trying to install record3d on my M1 Macbook Pro:

10.13-universal2-cpython-313', '-DPYTHON_EXECUTABLE=/Library/Frameworks/Python.framework/Versions/3.13/bin/python3', '-DCMAKE_BUILD_TYPE=Release', '-DBUILD_PYTHON_BINDINGS=ON']' returned non-zero exit status 1.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for record3d
Failed to build record3d
ERROR: Failed to build installable wheels for some pyproject.toml based projects (record3d)

What should I do to get it right?

marek-simonik, 2025-04-17:
Hello @ofergetz 

I think you copied only the end of the error log. The relevant information about the source of the error is located somewhere above the portion of the error log that you copied here.

Please paste the whole error log here so that I can help you fix the issue. If you do not want to copy the whole error log publicly, feel free to send me an email at support@record3d.app.

My first suggestion is that you try to execute the build command with `sudo`.

ofergetz, 2025-04-17:
sorry here it is: 
Ofers-MacBook-Pro-2:record3d ofergetz$ python3 -m pip install .  --user
Processing /Users/ofergetz/record3d
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting numpy (from record3d==1.4)
  Using cached numpy-2.2.4-cp313-cp313-macosx_14_0_arm64.whl.metadata (62 kB)
Using cached numpy-2.2.4-cp313-cp313-macosx_14_0_arm64.whl (5.1 MB)
Building wheels for collected packages: record3d
  Building wheel for record3d (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for record3d (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [147 lines of output]
      running bdist_wheel
      running build
      running build_ext
      -- The C compiler identification is AppleClang 15.0.0.15000309
      -- The CXX compiler identification is AppleClang 15.0.0.15000309
      -- Detecting C compiler ABI info
      -- Detecting C compiler ABI info - done
      -- Check for working C compiler: /usr/bin/cc - skipped
      -- Detecting C compile features
      -- Detecting C compile features - done
      -- Detecting CXX compiler ABI info
      -- Detecting CXX compiler ABI info - done
      -- Check for working CXX compiler: /usr/bin/c++ - skipped
      -- Detecting CXX compile features
      -- Detecting CXX compile features - done
      -- Performing Test CFLAG_Wall
      -- Performing Test CFLAG_Wall - Success
      -- Performing Test CFLAG_Wno_unknown_pragmas
      -- Performing Test CFLAG_Wno_unknown_pragmas - Success
      -- Performing Test CFLAG_Wno_unused_variable
      -- Performing Test CFLAG_Wno_unused_variable - Success
      -- Looking for stpncpy
      -- Looking for stpncpy - found
      -- Looking for sleep
      -- Looking for sleep - not found
      -- Looking for pselect
      -- Looking for pselect - found
      -- Looking for sys/inotify.h
      -- Looking for sys/inotify.h - not found
      CMake Warning at CMakeLists.txt:32 (find_package):
        By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
        asked CMake to find a package configuration file provided by "OpenCV", but
        CMake did not find one.
      
        Could not find a package configuration file provided by "OpenCV" with any
        of the following names:
      
          OpenCVConfig.cmake
          opencv-config.cmake
      
        Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
        "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
        provides a separate development package or SDK, be sure it has been
        installed.
      
      
      CMake Warning (dev) at python-bindings/pybind11/tools/FindPythonLibsNew.cmake:98 (find_package):
        Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
        are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
        the cmake_policy command to set the policy and suppress this warning.
      
      Call Stack (most recent call first):
        CMakeLists.txt:58 (include)
      This warning is for project developers.  Use -Wno-dev to suppress it.
[... 74 lines cut ...]
          cmd_obj.run()
          ~~~~~~~~~~~^^
        File "/private/var/folders/kr/4d632zps0jj75zdpnkz58dhm0000gn/T/pip-build-env-3y_2jgzx/overlay/lib/python3.13/site-packages/setuptools/_distutils/command/build.py", line 135, in run
          self.run_command(cmd_name)
          ~~~~~~~~~~~~~~~~^^^^^^^^^^
        File "/private/var/folders/kr/4d632zps0jj75zdpnkz58dhm0000gn/T/pip-build-env-3y_2jgzx/overlay/lib/python3.13/site-packages/setuptools/_distutils/cmd.py", line 357, in run_command
          self.distribution.run_command(command)
          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
        File "/private/var/folders/kr/4d632zps0jj75zdpnkz58dhm0000gn/T/pip-build-env-3y_2jgzx/overlay/lib/python3.13/site-packages/setuptools/dist.py", line 1104, in run_command
          super().run_command(command)
          ~~~~~~~~~~~~~~~~~~~^^^^^^^^^
        File "/private/var/folders/kr/4d632zps0jj75zdpnkz58dhm0000gn/T/pip-build-env-3y_2jgzx/overlay/lib/python3.13/site-packages/setuptools/_distutils/dist.py", line 1021, in run_command
          cmd_obj.run()
          ~~~~~~~~~~~^^
        File "<string>", line 34, in run
        File "<string>", line 59, in build_extension
        File "/Library/Frameworks/Python.framework/Versions/3.13/lib/python3.13/subprocess.py", line 419, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['cmake', '/Users/ofergetz/record3d', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/Users/ofergetz/record3d/build/lib.macosx-10.13-universal2-cpython-313', '-DPYTHON_EXECUTABLE=/Library/Frameworks/Python.framework/Versions/3.13/bin/python3', '-DCMAKE_BUILD_TYPE=Release', '-DBUILD_PYTHON_BINDINGS=ON']' returned non-zero exit status 1.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for record3d
Failed to build record3d
ERROR: Failed to build installable wheels for some pyproject.toml based projects (record3d)

marek-simonik, 2025-04-17:
Thanks, it seems like you have an old version of CMake (older than 3.5). Please try to install a newer CMake version and try to install the library again.

Since you are using Python version which is newer than 3.8, please patch the source code before as is explained in https://github.com/marek-simonik/record3d/issues/109#issuecomment-2763355826

marek-simonik, 2025-04-21:
@ofergetz I was able to replicate and fix the bug you described.

It seems like the "problem" was that you are using CMake 4, which likely introduced breaking changes that were incompatible with the version of `pybind11` currently used in `record3d`'s source code — thanks for spotting the bug and reporting it!

The solution is to replace the old `pybind11` version in `record3d`'s source code by the latest version of `pybind11` which is compatible with the latest versions of CMake.

I could successfully install `record3d` with CMake 4.0.1 by updating `pybind11` in the following way:

```bash
# Clone both the record3d library and the latest pybind11 library
git clone https://github.com/marek-simonik/record3d.git
git clone https://github.com/pybind/pybind11.git

# Remove the old pybind11 library's files from record3d's source code. The old version seems to be incompatible with CMake 4
rm -rf record3d/python-bindings/pybind11/*

# Copy the latest pybind11 version into record3d's source code
cp -r pybind11/include/ record3d/python-bindings/pybind11/
cp -r pybind11/tools/ record3d/python-bindings/pybind11/
cp pybind11/CMakeLists.txt record3d/python-bindings/pybind11/

# Install record3d
cd record3d/
python3 -m pip install . --user
```

marek-simonik, 2025-04-21:
@ofergetz I have just released a new version of the `record3d` library (version 1.4.1). You can now simply clone the `record3d` repo and install the library without applying any patches.