marek-simonik/record3d #80: issue with generating the .conf format

jeonhuhuhu, 2024-03-06:
I am using the Record3D App on my iPhone 14+ to collect `.r3d` data for a new environment. 

However, when reviewing other people's `.r3d` data, I noticed that the .conf format is generated and saved along with the data, but in my case, the `.conf` format is not being saved. 

(Left: my data, Right: general data with .conf format generated) 
[image]

Therefore, even though I looked into the settings and export options of the Record3D App, 
I couldn't find any functionality related to generating the .conf format.

[image]

Should I use an iPhone Pro series? Or should I adjust the shooting settings?

Could you please assist me with this? 

Thank you in advance for your help.

marek-simonik, 2024-03-06:
Hello,

the .conf files are seved automatically, but they are available only for 3D videos taken by LiDAR (which is available only on the iPhone Pro models, as you wrote).

The reason why LiDAR videos contain .conf files, is that Apple generates LiDAR depth maps by merging raw LiDAR measurements with color images using AI. Each pixels therefore gets a confidence score. Since the selfie FaceID camera does not use AI to generate depth maps (at least I hope so), there are no confidence maps available.

jeonhuhuhu, 2024-03-06:
> Hello,
> 
> the `.conf` files are saved automatically, but they are available only for 3D videos taken by LiDAR (which is available only on the iPhone Pro models, as you wrote).
> 
> The reason why LiDAR videos contain `.conf` files, is that Apple generates LiDAR depth maps by merging raw LiDAR measurements with color images using AI. Each pixel therefore gets a confidence score. Since the selfie FaceID camera does not use AI to generate depth maps (at least I hope so), there are no confidence maps available.

Thank you, I will test it with the iPhone Pro series!

jeonhuhuhu, 2024-03-07:
> Hello,
>
> the `.conf` files are saved automatically, but they are available only for 3D videos taken by LiDAR (which is available only on the iPhone Pro models, as you wrote).
>
> The reason why LiDAR [...]

I tested Record3D using my iPhone 14 Pro. 

However, I encountered an issue where the `.conf` format files are not generated intermittently. 

Is there any reason for this?

marek-simonik, 2024-03-07:
Can you please confirm that the test 3D video was recorded by the (rear) LiDAR camera and not by the selfie FaceID camera?

jeonhuhuhu, 2024-03-07:
> Can you please confirm that the test 3D video was recorded by the (rear) LiDAR camera and not by the selfie FaceID camera?

Hi!

Yes, we recorded using a (rear) LiDAR camera, and upon unpacking the `.r3d compression`, we noticed that some `.conf files` were missing intermittently. 

However, when recording in smaller spaces, the `.conf files` were generated properly without any omissions^^...

We're curious about the circumstances under which the `.conf format` information is missing from the recorded data.

marek-simonik, 2024-03-09:
Apologies for the delay in my reply; I tried to replicate the situation that you described, but I could not get Record3D to produce a `.depth` file without its corresponding `.conf` file.

Do I understand correctly that the issue has occured only when you recorded a 3D video in either a large enclosed area (e.g. in a gym or in a warehouse) or in a unbounded area (e.g. outside, in a park)? Does the 3D video has to be long (e.g. a few minutes) for the bug to occur or are the confidence map files missing even for short videos?

Are you able to reliably reproduce this problem or does it happen irregularly? If you *are* able to reproduce this problem, please let me know what specific steps should I take to attempt to reproduce the situation on my own.

Currently, the situation when the depth map is written to disk while at the same time the `.conf` file is not being written to the disk can happen either A) if the buffer holding the confidence map cannot be read from or B) if the API used for writing files to the disk throws an exception/error.

jeonhuhuhu, 2024-03-21:
> Apologies for the delay in my reply; I tried to replicate the situation that you described, but I could not get Record3D to produce a `.depth` file without its corresponding `.conf` file.
>
> Do I [...]

I apologize for the late reply. 

When I tested it, the `.conf` file was not created only once, and after that it was consistently created fine. 

This issue probably occurred because the first shot was not taken properly. 

We would like to proceed with a purchase to use the APP in earnest. 

However, what I'm curious about is whether it is possible to create a .conf file for all Pro versions, regardless of `Apple's version`. 

In other words, we have a variety of `iPad Pro models`, so we want to check them out before purchasing the app. 

I'm curious if it only works on a specific Pro version or if it's iOS 16 or higher. 

Thank you in advance for your kind response

marek-simonik, 2024-03-25:
> However, what I'm curious about is whether it is possible to create a .conf file for all Pro versions, regardless of `Apple's version`.

I am not sure what you meant by `Apple's version`, but in general, the `.conf` files are going to be generated for all LiDAR videos, therefore if you need `.conf` files, then you will need an iPad Pro (or an iPhone Pro) which has LiDAR (the first iPad Pro with LiDAR sensor was released in 2020). At the same time, Record3D requires iOS 16 or newer.

You can easily check whether an iPad has LiDAR by downloading Record3D for free, opening the app and checking if there is the camera-switch button on the right side of the big red circular toggle button in the "Record" tab of Record3D; if there is the camera-switch button, then `.conf` file are going to be generated in LiDAR mode.

Note that you do **not** need to buy Record3D to perform this test; moreover, you can record 3 videos to evaluate that the `.conf` files are indeed going to be generated.

In case I forgot to answer any of your questions or if you have follow-up questions, please let me know.