marek-simonik/record3d #81: How to access conf values when streaming?

alik-git, 2024-04-20:
Hi there, thanks for this awesome repo!

Is it possible to access the conf values for depth when streaming? or is that only possible when saving the r3d file?

Please let me know, thanks!

marek-simonik, 2024-04-22:
Hi,

the depth confidence files are currently available only for the recorded sequences (in `.r3d` files), but adding the confidence maps also to the USB stream is on my TODO list for a future version of Record3D.

marek-simonik, 2024-05-28:
This feature has been implemented in Record3D 1.10, which necessitated breaking changes to be done to the `record3d` library. Please update to Record3D 1.10 (or newer) and also update to `record3d` library version 1.4.

You can access confidence maps as shown here: https://github.com/marek-simonik/record3d/blob/0c6e7692da100aad2d5195e307f01e75427a6a2a/demo-main.py#L52

A confidence map is a `uint8` image, where each pixel can have one of the following values representing the confidence of estimated depth value in the corresponding depth map:
  - `0` for low-confidence,
  - `1` for medium-confidence,
  - and `2` for high-confidence depth estimate.

Note that confidence maps are available only for LiDAR streams (otherwise the confidence map has resolution `0x0` px).

closed 2024-05-28 by marek-simonik

jeonhuhuhu, 2024-09-26:
> This feature has been implemented in Record3D 1.10, which necessitated breaking changes to be done to the `record3d` library. Please update to Record3D 1.10 (or newer) and also update to `record3d` [...]

How can I run demo-main.py from the answer above while simultaneously saving the .jpg, .depth, and .conf values ​​for individual frames and finally saving them in .r3d data format format along with Metadata?

marek-simonik, 2024-09-28:
> How can I run demo-main.py from the answer above while simultaneously saving the .jpg, .depth, and .conf values ​​for individual frames and finally saving them in .r3d data format format along with Metadata?

I am not sure if I understood your question, but I think you will want to modify the `demo-main.py` file so that in the `while` loop it saves all the necessary per-frame data and metadata to memory and at the end of the stream, you construct the `.r3d` file out of the data. See my answer here for more details: https://github.com/marek-simonik/record3d/issues/93