marek-simonik/record3d #84: libc++abi: terminating due to uncaught exception of type std::bad_alloc: std::bad_alloc

takeyamayuki, 2024-05-29:
Thanks for a good system!

An error occurred with the recent update ...
```
❯ python demo-main.py
Searching for devices
1 device(s) found
        ID: 4776
        UDID: 00008110-000231E60242401E

libc++abi: terminating due to uncaught exception of type std::bad_alloc: std::bad_alloc
[1]    4247 abort      python demo-main.py
```

The environment is .
- record3d=1.4
- Record3D on iOS is 1.10
- Python 3.10.11

takeyamayuki, 2024-05-29:
Alternatively, following error is output:

```
❯ python demo-main.py
Searching for devices
1 device(s) found
        ID: 4776
        UDID: 00008110-000231E60242401E

libc++abi: terminating due to uncaught exception of type std::length_error: vector
[1]    5069 abort      python demo-main.py
```

takeyamayuki, 2024-05-29:
Uploading iOS to 17.5.1 fixed it.

marek-simonik, 2024-05-29:
I apologize for this bug. Unfortunately, I could not fully replicate your setup nor the crash. Although I'm happy to hear that updating to iOS 17.5.1 fixed the problem, I still wonder what might have caused the it… If I will manage to get access to an iPhone with iOS 16, I will investigate this further.

marek-simonik, 2024-05-29:
I have just tested what would happen if I tried to stream with `record3d` **v1.4** *(the library)* while running Record3D **v1.9** *(the iOS app)*.

In this setup, I was able to replicate the same error that you experienced (I believe this specific error occurred because the size and structure of the header containing byte sizes of images and other metadata has changed after the update, so the new `record3d` library attempted to resize an `std::vector` to a large capacity because it incorrectly assumed that it read a size of an image buffer while in reality, it probably read a pixel's color value);

```shell
libc++abi: terminating due to uncaught exception of type std::bad_alloc: std::bad_alloc
Abort trap: 6
```

May I ask if you checked that you were running Record3D 1.10 the first time you opened Record3D after finishing the update (you can check what version of Record3D you are running at the bottom of the Settings tab)?

It has happened to me a couple of times that after opening an (seemingly freshly updated) iOS app, I was still running the old version of that app and I had to re-download the update from the App Store to truly update to the latest version (possible iOS bug?). Perhaps this might have happened in your case too?

I will try to test the new Record3D update on a device running iOS 16 too though.