marek-simonik/record3d #87: origin of camera pose

junbangliang, 2024-07-12:
Hi,

Thank you for the great work. I am new to the Apple ARkit I am trying to use the data from the app for robotics. In this case I would like to know the precise origin where the local coordinate system is when we obtain pose from self.session.get_camera_pose(). It is origin at the lidar sensor or one of the rgb sensor?

Thanks!

marek-simonik, 2024-07-18:
Hello,

thank you for using Record3D. Unfortunately, I am not sure with absolute certainty where the origin is located relative to the device's hardware. The diagram in [this article](https://developer.apple.com/documentation/arkit/arconfiguration/worldalignment/gravity) displays the origin to be in the center of the device, but there is NO explicit mention of the location of the origin point in the written text, so I would not rely on the diagram alone (it might be there just for illustration purposes).

In my understanding, the documentation is not clear on the exact position of the origin relative to the device's body. If you would also not be able to find more details in the documentation, then I recommend to ask Apple Developer Support for clarification of documentation on this topic.

junbangliang, 2024-07-23:
Thank you for your help!

closed 2024-07-23 by junbangliang