marek-simonik/record3d #90: Problem starting stream with demo-main.py

lukas486, 2024-08-24:
Hello Marek, 

thanks a lot for your work. 
Sadly, I am not able to start the stream through your demo-main.py code. The error I get is: 
```
Traceback (most recent call last):
  File "C:\Users\finkpielsticker\record3d\demo-main.py", line 82, in <module>
    app.start_processing_stream()
  File "C:\Users\finkpielsticker\record3d\demo-main.py", line 50, in start_processing_stream
    depth = self.session.get_depth_frame()
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: resize only works on single-segment arrays
```
If I start streaming on the app and later start the script it shows this error right away. If I start the script and after press on record in the app it pops up then only. 
I've tried a few things I saw here in similar issues, but no success. Maybe you have an idea what is happening? 
Thanks a lot again! 
Currently running: Windows 10 Pro 22H2; iOS 17.6.1; Record3D 1.10.2; Python 3.12
All the best,
Lukas

marek-simonik, 2024-08-24:
Hello Lukas,

thanks for using Record3D. This issue sounds remotely similar to https://github.com/marek-simonik/record3d/issues/89#issuecomment-2294888450 — i.e. please try one of these options:
- use Python 3.8
- or apply the patch mentioned in the [comment](https://github.com/marek-simonik/record3d/issues/89#issuecomment-2294888450), recompile/reinstall such modified source code.

Please let me know if either of the two solutions helped.

Thank you,
Marek

lukas486, 2024-08-24:
Hello Marek, 
thanks for the quick reply and your solution. 
I reinstalled python as  version 3.8 and all other packages. It works! 

All the best!
Lukas

closed 2024-08-24 by lukas486