marek-simonik/record3d #98: USB streaming does not work on latest versions

johnpayne-gc, 2024-10-21:
I am running the latest Record3d app and library versions, and this is the error I get when running `demo-main.py`:

```
Searching for devices
1 device(s) found
	ID: 4776
	UDID: 00008120-001430AA1E80201E

Traceback (most recent call last):
  File "/Users/johnpayne/demo-main.py", line 82, in <module>
    app.start_processing_stream()
  File "/Users/johnpayne/demo-main.py", line 50, in start_processing_stream
    depth = self.session.get_depth_frame()
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: resize only works on single-segment arrays
```

marek-simonik, 2024-10-22:
I think the issue might be related to newer versions of Python (Python 3.10+). Try either using an older Python version or test if the fix suggested [in this comment](https://github.com/marek-simonik/record3d/issues/89#issuecomment-2294888450) works for you when using your current Python version. Please let me know if either helped. Thanks.

sleeplessTLV, 2024-10-22:
Hi, I tried to follow these instructions but the sequence is just not clear:
1. I changed the relevant lines in Record3DStream.h (see attached file)
2. I added #define Pyhton_Bindings_Build to the start since without it the whole code change does not take effect - was I wrong in that?
3. I ran cmake -DCMAKE_BUILD_TYPE=Release and it executed without any errors
4. However when I run with python3.12 I get exactly the same error as before :

Searching for devices
1 device(s) found
        ID: 4776
        UDID: 00008120-000265102184C01E

Traceback (most recent call last):
  File "D:\ProgramData\Lidar\record3d\demo-main.py", line 82, in <module>
    app.start_processing_stream()
  File "D:\ProgramData\Lidar\record3d\demo-main.py", line 50, in start_processing_stream
    depth = self.session.get_depth_frame()
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: resize only works on single-segment arrays

If I try an older python version (3.8) I get

D:\ProgramData\Lidar\record3d>python3.8 demo-main.py
Traceback (most recent call last):
  File "demo-main.py", line 2, in <module>
    from record3d import Record3DStream
ModuleNotFoundError: No module named 'record3d'

sleeplessTLV, 2024-10-22:
Hi, regardless of Python issues which I still have with the update, I compiled the demo.exe app. it runs with no error but nothing shows on the screen:
D:\ProgramData\Lidar\record3d\Debug>demo.exe
Found 1 iOS device(s):
        Device ID: 4776
        UDID: 00008120-000265102184C01E

Trying to connect to device with ID 4776.
Connected and starting to stream. Enable USB streaming in the Record3D iOS app (https://record3d.app/) in case you don't see RGBD stream.
Stream stopped!

johnpayne-gc, 2024-10-22:
Sorry, don't know how I missed that comment. FYI it doesn't work with 3.9 either, 3.8 is the latest version I could get it to work on.

Closing this issue since it is a duplicate.

closed 2024-10-22 by johnpayne-gc