marek-simonik/record3d #107: Misalignment of poses when using Open3D to generate point cloud

EricLiangOS, 2025-01-21:
I am currently using the depth images, rgb images, and camera poses recorded by Record3D in order to generate a point cloud for each frame of the video. The scale of the point cloud looks correct, but using the poses from the metadata results in misalignment between the points, which is very different compared to the generated point clouds by the app.

Is there an alignment process to produce the results from the app?

Below is a snippet of code for creating the point cloud

```
    with open(metadata_path, 'r') as file:
        metadata_dict = json.load(file)

    poses_data = np.array(metadata_dict["poses"])
    
    camera_to_worlds = np.concatenate(
        [Rotation.from_quat(poses_data[:, :4]).as_matrix(), poses_data[:, 4:, None]],
        axis=-1,
    ).astype(np.float32)

    homogeneous_coord = np.zeros_like(camera_to_worlds[..., :1, :])
    homogeneous_coord[..., :, 3] = 1
    camera_to_worlds = np.concatenate([camera_to_worlds, homogeneous_coord], -2)

    K = np.array(metadata_dict["K"]).reshape((3, 3)).T
    focal_length = K[0, 0]

    intrinsics = o3d.camera.PinholeCameraIntrinsic(
        width=W, 
        height=H,
        fx=focal_length,  # focal length x
        fy=focal_length, 
        cx=W/2, 
        cy=H/2 
    )
```

```
    depth_img[~alpha_mask] = 0

    rgbd_image = o3d.geometry.RGBDImage.create_from_color_and_depth(
        o3d.geometry.Image(mask_img[:, :, :3].astype(np.uint8)),
        o3d.geometry.Image(depth_img),
        depth_scale=1.0,
        depth_trunc=1000.0,
        convert_rgb_to_intensity=False
    )

    pcd = o3d.geometry.PointCloud.create_from_rgbd_image(
        rgbd_image, intrinsics, extrinsics
    )
```

The following is the misalignment from the RGBD

[image]

closed 2025-01-21 by EricLiangOS

dcedgren, 2025-02-27:
I wonder if you could share your code that ended up working for determining the 4x4 extrinsics matrix? I'm also struggling with this problem.

dcedgren, 2025-03-07:
Here's what ended up working for me. I was getting tripped up by the different conventions for the Y and Z axes between an image array and a point cloud. After I flipped these two axes it worked correctly and my point clouds successfully align. Hopefully this will benefit someone else!

`import cv2
import liblzfse
import json
import numpy as np
import open3d as o3d

def load_depth(filepath):
    with open(filepath, 'rb') as depth_fh:
        raw_bytes = depth_fh.read()
        decompressed_bytes = liblzfse.decompress(raw_bytes)
        depth_img = np.frombuffer(decompressed_bytes, dtype=np.float32)
        depth_img = depth_img.reshape((256, 192)) 
    return depth_img

# Read metadata
with open(root_path.joinpath('metadata'), "rb") as f:
    metadata = json.loads(f.read())

# Create an empty point cloud to store the unified result
pcd = o3d.geometry.PointCloud()

num_frames = len(metadata['frameTimestamps'])
frame_interval = 30
for frame_num in range(num_frames):
    if frame_num % frame_interval == 0:
        depth_filepath = root_path.joinpath(f'rgbd/{frame_num}.depth')
        rgb_filepath = root_path.joinpath(f'rgbd/{frame_num}.jpg')

        rgb = cv2.imread(str(rgb_filepath))
        rgb = cv2.cvtColor(rgb, cv2.COLOR_BGR2RGB)
        rgb = cv2.resize(rgb, (192, 256), cv2.INTER_NEAREST_EXACT)
        
        depth = load_depth(str(depth_filepath))

        # Create intrinsic matrix
        frame_K = metadata['perFrameIntrinsicCoeffs'][frame_num]
        # Because rgb image was resized intrinsic matrix also needs to be
        scale = 256/960
        frame_K = [i*scale for i in frame_K]
        intrinsic = o3d.camera.PinholeCameraIntrinsic(width=192, height=256, fx=frame_K[0], fy=frame_K[1], cx=frame_K[2], cy=frame_K[3])

        # Create 4x4 extrinsic matrix
        qx, qy, qz, qw, tx, ty, tz = metadata['poses'][frame_num]
        extrinsic = np.eye(4)
        extrinsic[:3, :3] = o3d.geometry.get_rotation_matrix_from_quaternion(np.array([qw, qx, qy, qz]))
        extrinsic[:3, 3] = [tx, ty, tz]

        # Create an RGBD image
        rgbd = o3d.geometry.RGBDImage.create_from_color_and_depth(
            o3d.geometry.Image(rgb),
            o3d.geometry.Image(depth),
            depth_scale=1.0,
            depth_trunc=10.0,     # Maximum depth in meters
            convert_rgb_to_intensity=False,
        )
        
        # Create point cloud from the RGBD image in camera space
        this_pcd = o3d.geometry.PointCloud.create_from_rgbd_image(
            rgbd, intrinsic,
        )
        # Flip y and z axes because image and point cloud have different conventions
        flip_transform = [[1, 0, 0, 0], [0, -1, 0, 0], [0, 0, -1, 0], [0, 0, 0, 1]]
        this_pcd.transform(flip_transform)
        
        # Transform from camera space to world space
        this_pcd.transform(extrinsic)

        # Add points to unified point cloud
        pcd += this_pcd

o3d.visualization.draw_geometries([pcd])
`