marek-simonik/record3d #111: White marker toggle while recording

sadanand1120, 2025-04-12:
Hi, thanks for the awesome app. While recording there is a white line-marker like toggle. I am not sure whats that doing and what should it be set to. I guess (?) something to do with depth? I want max accurate depth beyond 3.0 meters. What do I set it to?

Thanks,

marek-simonik, 2025-04-13:
Hi,

thanks for using Record3D :)!

Do you mean the horizontal slider, which is located between the red circular toggle button and the live camera view (please see the attached image)? This is the depth range slider.

If you want Record3D to save depth maps without clamping the depth values, then set the slider to its maximum value (like in the attached image). This will save the original, unmodified depth values.

In case you are exporting the 3D videos into the RGBD mp4 format and want to increase the maximum depth that is being encoded into color, then go to the Settings tab > Export options > enable "RGBD mp4 video dynamic max. depth". But I guess this is not what you asked for.

Unfortunately, the depth maps provided by Apple will NOT be accurate for points beyond 3 meters due to noise (this is true both for LiDAR and for FaceID videos).

[image]

sadanand1120, 2025-04-14:
Got it. Thanks for the clarification!

closed 2025-04-14 by sadanand1120