marek-simonik/record3d #115: How to get a higher resolution of the RGB images

byran-wang, 2025-04-14:
Hi @marek-simonik,

As suggested in [#114](https://github.com/marek-simonik/record3d/issues/114), I enabled "Increased selfie camera resolution", but the exported RGB images still have a low resolution (480×640) when using the 'EXR + JPG sequence' option. I tested this on both an iPhone 11 Pro and 13 Pro. Any idea why i can not get a  higher resolution of the RGB images? Thanks.

marek-simonik, 2025-04-14:
This sounds like a bug — please send me an email at support@record3d.app and I will let you test a fixed version of the app. Thanks!

marek-simonik, 2025-04-16:
I have not received your email; it is not in my Spam folder either (perhaps my email provider filtered it?). But I have just sent you an email at the address showed in the screenshot you shared.

If you have a secondary email address, please try to contact me from it.

Please let me know if you received my email. Thanks.

byran-wang, 2025-04-18:
I've received the email.

I’ve tested the Beta version of Record3D and the RGB image resolution of FaceID has been upgraded to 1440×1920 (high resolution) which is what I want.

Thanks.

closed 2025-04-18 by byran-wang