marek-simonik/record3d #117: No image when streaming USB to Unbuntu 22.04

sven1729, 2025-04-16:
I have pip installed the record3d library into a python virtual environment.

I'm on Ubuntu 22.04. With following libraries:
opencv 4.11.0.86
numpy 2.2.4
record3d 1.4
python 3.10.14

Here is the setup:
- An iphone 12 pro plugged in via USB, set to USB streaming. 
- start the app and press the red record button.
- launch the `python demo-main.py` script.
- Script prints 1 device found and phone show 'Connected, streaming RGBD video'
- Script appears stuck on line 47, `self.event.wait()`, no frame is displayed.

Can someone suggest what is setup and app settings are required to get a minimum working demo and receive RGBD frames?

marek-simonik, 2025-04-16:
The setup you followed seems OK to me.

Did you see the ["*Trust your computer*"](https://support.apple.com/en-us/109054) message on your iPhone and pressed "Trust"? (I assume you did.)

Please try this:

1. Unplug the USB cable from your iPhone.
2. [Force-restart](https://support.apple.com/guide/iphone/force-restart-iphone-iph8903c3ee6/ios) the iPhone.
3. Unlock your iPhone.
4. Connect it to the PC via USB.
5. Try to stream again (by following the steps you mentioned in your post).

There is one more thing you need to know: since you are using Python 3.10, you will need to patch the source code of the `record3d` library and reinstall it from source — see https://github.com/marek-simonik/record3d/issues/109#issuecomment-2763355826 for more details.

sven1729, 2025-04-16:
Thanks. I have made the change but now an earlier error appears when I start the script.

```
Searching for devices
1 device(s) found
	ID: 4776
	UDID: XXXXXXXXXXXXXXXXXXXXXXXXX

Traceback (most recent call last):
  File "/home/XXXXXXXXX/record3d/record3d/demo-main.py", line 82, in <module>
    app.start_processing_stream()
  File "/home/XXXXXXXXX/record3d/record3d/demo-main.py", line 50, in start_processing_stream
    depth = self.session.get_depth_frame()
ValueError: resize only works on single-segment arrays
```

I have 

opencv 2.2.4
numpy 4.11.0.86
record3d 1.4 compiled from source with patch.

Any thoughts?

marek-simonik, 2025-04-17:
That is weird — this is exactly the error, which the patch aims to fix. Can you please double-check that you are using the patched version of the library? Thanks.